madhephaestus/WiiChuck #3: Classic controller

20161220-rootfest, 2017-06-17:
how can i read the value of the left joystick on classic controller ? 

Serial.println joyYLeftBytes();

is not working

madhephaestus, 2017-06-17:
controllerObject.getJoyYLeft() is the function. Make sure you are using the object, it's not a static method.

20161220-rootfest, 2017-06-17:
Sorry but i dont get it to work even with

Serial.println controllerObject.getJoyYLeft();

madhephaestus, 2017-06-17:
What did you name your controller object? Are you familiar with object oriented programming?  Is there a reason your example is written without parentheses? Did the classic controller example work?

20161220-rootfest, 2017-06-18:
> What did you name your controller object? 

I named it classic.getJoyYLeft(); if this is correct. 

> Are you familiar with object oriented programming?

No im not familiar with object oriented programming.

> Is there a reason your example is written without parentheses?

It dinnt work with them for me but i think i just did that wrong . Can you help me ? 

> Did the classic controller example work?

Yes

madhephaestus, 2017-06-18:
Modify the example and in loop put two new lines:
Serial.print( "\n Joy y left value =");
Serial.print( classic.getJoyYLeft());

20161220-rootfest, 2017-06-19:
Thank you but this is  only working half ways it only prints the first state and then repeat it no changing values :( 

Starting WiiChuck Generic Controller Demo
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31
 Joy y left value =31

fraser125, 2017-08-02:
I think you are looking for a simple complete example of using this library for Classic Controller:
```
#include <WiiChuck.h>

#if defined(ARDUINO_ARCH_ESP8266)
#define SDA D2
#define SCL D1
#endif

Classic classic(SDA, SCL);
 
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  classic.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  classic.readData(); 
  int joyLeftY = classic.getJoyYLeft();
  Serial.println(joyLeftY, DEC);
}
```

The value was repeating because you weren't calling 'classic.readData(); inside the loop().

madhephaestus, 2017-08-02:
Update to 0.0.21 and try the new Classic_Simple example

closed 2017-08-02 by madhephaestus