madhephaestus/WiiChuck #8: Detach Fork?

dmadison, 2017-10-16:
It looks like this project was forked from a repo that's just a collection of various libraries. It might be prudent to contact GitHub support and have the fork detached so that it can be developed independently.

madhephaestus, 2017-10-16:
Yeah.... I turned on the issues and this is the Main forf for WiiChuck, but i like the attribution of the fork back to the original curator that pulled together the header files. I dunno, its more about "credit where credit is due" in my mind.

closed 2017-10-16 by madhephaestus

dmadison, 2017-10-16:
Fair enough! It's just a little strange seeing the commit comparison at the top of the repo with how the original is set up. You might consider detaching it and adding a note in the `readme` at some point in the future.

madhephaestus, 2017-10-16:
I added an explanation to the Readme. Thanks for contributing to make WiiChuck even better!