madhephaestus/WiiChuck #26: Incompatible with Teensy?

xoxota99, 2018-01-19:
I'm getting an error at compile time, due to a reference to TWCR, a register that doesn't exist in Teensyduino.

```
src/Accessory.cpp: In static member function 'static void Accessory::switchMult
iplexer(uint8_t, uint8_t)':
src/Accessory.cpp:94:8: error: 'TWCR' was not declared in this scope
if(TWCR == 0){ Wire.begin(); } // Start I2C if it's not running
^
```

madhephaestus, 2018-01-19:
which teensy? this is tested working on the 3.2 and 3.5

xoxota99, 2018-01-19:
Ah, maybe Teensy is a red herring then? I'm using Teensy 3.2, via PlatformIO. Can repro using the example apps.

madhephaestus, 2018-01-20:
I just tested in Arduino 1.8 on the 3.2 using version 0.0.20, must be a bug in 23, i see the error now. looking into it. In arduino just set the version number to 0.0.20 and is should work