madhephaestus/WiiChuck #29: imposible use with esp32 :(

inukisoft, 2018-02-12:
Hi, I am testing the sample for wiii nunchakus (from examples) in esp32 dev, but ...

`
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction)
. Exception was unhandled.
Register dump:
PC      : 0x400d0a60  PS      : 0x00060930  A0      : 0x800e676a  A1      : 0x3ffca5d0  
A2      : 0x3ffc2234  A3      : 0x3ffc23e4  A4      : 0x3ffc5524  A5      : 0x000000ff  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d0a5e  A9      : 0x3ffca5b0  
A10     : 0x00000044  A11     : 0x3ffc5524  A12     : 0x000000c8  A13     : 0x00000080  
A14     : 0x0000000a  A15     : 0x3ffc5534  SAR     : 0x0000001a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d33d4  LEND    : 0x400d33f3  LCOUNT  : 0x00000000  

Backtrace: 0x400d0a60:0x3ffca5d0 0x400e6767:0x3ffca5f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Starting WiiChuck Nunchuck Controller Demo

`
(reeboot automatically and the message appear again) 

I am checked the git history.  You revert support with esp32 (problem with servo), but not work for me.  In last versión you include <Wire.h> but it without initialice (21,22).

madhephaestus, 2018-02-12:
Im looking into this today, just got it working with the latest commit https://github.com/madhephaestus/WiiChuck/tree/409270eb500dc5ce4d096e9cee6e86ec1b6abdc9

Let me know if that works for you?

madhephaestus, 2018-02-15:
I made a release, this should be working on the esp32 now.

closed 2018-02-15 by madhephaestus

inukisoft, 2018-02-15:
Funciona! 

It's ok, thanks a lot.