madhephaestus/WiiChuck #38: Using the Wii nunchuck with Makeblock Ultimate 2.0

clplaneguy, 2019-03-04:
I have the Makeblock Ultimate 2.0 robot kit.  It uses an Arduino Mega 2560 (like or clone?) with a Bluetooth module.  Makeblock has its own software to remote control the robot by Bluetooth.  I have also purchased the wireless Bluetooth remote. Independent of the app, the board reads the controller commands, but I do not know what they mean and Makeblock considers them proprietary and does not give them out.  You have a library for the Nintendo WII nunchuck.  I would again think that the communication protocol would be proprietary and unavailable.  However, they seem to be in the public domain.  Robert Eisele has worked on this issue in the past.  He has a web page, pictures, instructions,  wiring diagrams and a processing sketch.  So, I used his info to wire my nunchuck.  But Arduino cannot find his library even if I put it in both the Arduino application and Arduino sketch folders.  With your library, I went into the Arduino Library Manager and installed.  It runs the sample program.  But it prints so much that I can not read the content. If the nunchuck is not plugged in I can barely read the word ERROR.  With the nunchuck plugged in, I think the numbers never change.

Part of the problem may be the connection between the Arduino and the nunchuck.  The nunchuck is NOT plugged directly into the main board.  The Makeblock Ultimate 2.0 has a HAT with three RJ25 ports  
(6, 7 & 8).  They are labeled SCL, SDA, GND, 5V, pin#, and pin#.  So, they seem to be I²C (I2C).  And in fact, I do get a response when plugging-in and un-plugging the nunchuck.  Eisele states that the nunchuck is 3.3v and not 5v.  So, I use an LED or drop the voltage(?).  Although I have not located my meter yet.  

So, I am wondering what my next step is.  Do you have any ideas?

About Robert Eisele and this blog
https://github.com/infusion/Fritzing/tree/master/Nunchuk     
Using a Wii Nunchuk with Arduino
December 19th, 2016.
# Wii-Nunchuk
[image]
Schema
---
[image]
Documentation
---
The full documentation can be found [here](http://www.xarg.org/2016/12/using-a-wii-nunchuk-with-arduino/).
Last access 6 months ago.
Has webpage and picture instructions.
Dot h library does not work even if in both the Arduino application and Arduino schetch folders.

Kevin Harrington
# WiiChuck
An Arduino library for talking to every extension controller made for the Wii including the Nunchuk and Classic Controller over I²C.
## Supported Controllers:
* Nunchuk
* Classic Controller
* Guitar Hero Guitar
* Guitar Hero Drums
* DJ Hero
* Drawesome Tablet
* Taiko Drums
# Repository Structure 
This repository is forked from a curated set of old Arduino Libraries. I kept the old commits and the fork linking to keep attribution to the work done before I picked up the torch. In my mind we all see farther by standing on the shoulders of giants, so it is only proper to give credit where credit is due.
https://github.com/madhephaestus/WiiChuck
Accessed throuth Arduino application, Runs, prints to screen. Examples.

madhephaestus, 2019-03-04:
The Wii nunchuck uses 3.3 or 5v without modification. Since the Arduino is
5v io, then you will have to be in 5v mode on the i2c. You might also need
to add a the 4.7k ohm pull up resistors on SDA and SCL. Also make sure you
have the version I published yesterday, 0.3.1 has a fix for the generic
nunchucks.

clplaneguy, 2019-03-04:
I removed the LED, converting to a full 5v.  I also have a CanaKit Raspberry Pi 3 B+ kit with 10k ohm resistors.  So, I used those for pull-up.  With a delay(1000), I can read all the values.  And I do have changing values.  (JoyX, JoyY) and a response to the C and Z buttons.

So, I have reached this milestone.  Actually, what I a striving for is wireless control.  I still have to purchase the wireless controller.  Then I am expecting more difficulty to get that to work.

**THANK YOU!**

closed 2019-03-04 by madhephaestus

madhephaestus, 2019-03-04:
You may be interested in my other library EspWii. It lets you take 2 ESP32's, plug a Wii accessory into one, and use the other as an RC controlled robot. 

https://github.com/madhephaestus/EspWii